borevitzlab/spc-eyepi #23: Need way to determine which camera is currently plugged into pi

TimeScience, 2015-01-22:
* Hard to tell which camera is currently plugged into the pi. 

* To start, the landing page should draw a square or light up a green button or something for configs of each detected camera(s). 

* Also ideally one could hit the "take picture" button to make a camera take a picture to identify itself. 
* I know the shorter serial ID is available in the config page but that's covered by the camera tripod mount so not actually that accessible in practice.

gdunstone, 2015-01-23:
> Assigned #23 <https://github.com/borevitzlab/spc-eyepi/issues/23> to
> @gdunstone <https://github.com/gdunstone>.

I'm aware that the serial is covered.

I'll look into doing that.

gdunstone, 2015-02-24:
Maybe we could tape the serial number for each camera to the top?
Like the raw one (thats all in hex)

gdunstone, 2015-04-15:
Will add a take_preview function that waits for the camera.

gdunstone, 2015-04-15:
need to wait for locking of the camera

gdunstone, 2015-04-16:
The cameras now adequately lock themselves.
As of update 2.0.5-37 it is stable with camera locking.

closed 2015-10-17 by gdunstone