borevitzlab/spc-eyepi #24: UI improvements for the camera settings page

TimeScience, 2015-01-23:
[image]
see photo :)

gdunstone, 2015-01-23:
> Assigned #24 <https://github.com/borevitzlab/spc-eyepi/issues/24> to
> @gdunstone <https://github.com/gdunstone>.

The ftp settings ARENT global. Each camera has their own uploader.

gdunstone, 2015-01-23:
And ill change the background writing.

gdunstone, 2015-02-24:
background writing changed.
advanced options hidden.

gdunstone, 2015-02-24:
Cant put the long camera id up there because otherwise what to name other cameras that dont have an EOS serial number (like for any canon camera that isnt an EOS

gdunstone, 2015-02-24:
closing and reopening as remaining issue.

Added feature to help the user with camera naming:
Config button in images page.

closed 2015-02-24 by gdunstone