borevitzlab/spc-eyepi #27: Camera stat web page and New configuration for ipaddress.html file

TimeScience, 2015-03-16:
To make it easier to manage cameras, can you change the html file to include the following
* New filename: camInfo.html
* Info to include
  * For each camera, make a div box with the following (assume 2 DSLR+ picam, so 3 div's in a row):
    * Camera Name
    * Is camera enabled or not
    * Date of last good capture 
    * Camera IP address as link to config GUI
    * Number of images remaining to upload (so we can track if there is a backup)
    * Embed of last good image (small version if possible, if not I'll write a server side script to downsize them)
    * Page could look something like below (but with only 3 cameras).

* Can you then write a wrapper HTML page that iFrames each of the pi html pages. There should be 7 frames, one for each pi in each chamber. So the page will have 7 rows Starting with chamber 2 (so 2,3,4,5,35,36,37). Goal is to be able to load a singe page that shows status of all cameras. Hard coding thing is fine if it happens faster than any snazzy coding.
   * When page loads and there is no existing html page for that chamber/camera it can just 404, no need for extra coding

[image]

gdunstone, 2015-03-25:
I need the picam directories, to be shared on the internet then.

gdunstone, 2015-04-07:
So those picam directories, they are shared. but windows is stripping the permissions off my files when i upload them.
Unfortunately I cant change this from the picam side, we need some way of setting the read permissions of the camupload folder (and, recursively, its subfolders) on the windows server

gdunstone, 2015-04-15:
This has almost all been sorted out.

The cameras now upload a JSON with their metadata.

And that page is

closed 2015-04-15 by gdunstone